jeffeb3/v1pi #2: Firefox scrambled display in Octoprint mode

billsey, 2018-05-05:
[image]

I haven't got it all setup yet, just fired it up yesterday for a while, then installed the camera today. A page refresh on this times out, a second refresh gave me this again.

jeffeb3, 2018-05-06:
I don't think I have the https version working right. Is it possible you are connected to the https://whatever?.

That looks like the main file is loading fine, but the supporting fold aren't.

billsey, 2018-05-08:
I'm thinking it's timeouts when loading the page, the first part gets through but it eventually fails due to the delay. I fired up a second Pi with the same configuration and I'm not even getting that one on the network at all. I wonder if they can conflict with each other when operating too closely together? They're both from Canakit and the first one I purchased from them was fine.

jeffeb3, 2018-05-09:
I think they will both have the same hostname (v1pi) and that will cause them to act funny. You can change them with raspiconfig or change it in /etc/hostname and /etc/hosts.

jeffeb3, 2019-05-02:
Bill, did you ever get this one figured out? Did the second hostname fix it?

jeffeb3, 2019-12-01:
Hey Bill, Do you have any more info on this? Have you been using v1pi, or did you just give up? I haven't seen anything like this in my testing.

billsey, 2019-12-04:
I haven't been actually using it. When I get home this weekend I'll fire up that pi and install the new one to see if the symptoms have changed.

Bill

Sent from my Verizon, Samsung Galaxy smartphone