jeffeb3/v1pi #3: Server Crashing - CNC.js

leitenky, 2019-02-12:
After about 5 minutes CNC..js seems to crash. I am using your image with no additional configuration and after about 5 minutes the server seems to crash and require a restart. Ive tried creating the image multiple times, verified the sd card is good, and its a new Pi3B+

Is there somewhere the logs are kept?

jeffeb3, 2019-02-12:
Bummer. I've been spoiled since it's been so reliable.

It's running from systemd. To see the logs you can use journalctl

`journalctl -u cncjs.SERVICE`

Add a '-b' to only show one processes logs. 

I was just trying this and I didn't see any output, but mine has been working fine. But to get more you can edit /etc/default/cncjs and add -vvv to the DAEMON_ARGS variable.

leitenky, 2019-02-12:
I tried adding -vvv to the Daemon_args and when I try journalctl -u CNCJs.SERVICE I get --no enteries--

Any other advice would be appreciated!

leitenky, 2019-02-12:
I can see after CNCjs starts that the ram slowly Kris fishing and after a couple minutes goes back down to approximately 10%.  At that point CNCJs server seems to crash

jeffeb3, 2019-02-12:
1) It is case sensitive, and I'm not sure why I suggested SERVICE in the first place. I blame my phone, sorry. It should be:
`journalctl -u cncjs.service`
2) You can check to see if it's running with `sudo service cncjs status`
3) If it's stopped, you can run it manually by doing this:
```
sudo su # become root
cnc -vvv
```

jeffeb3, 2019-02-12:
Do you have any ideas why it might be working harder or be different on your pi?

leitenky, 2019-02-12:
No that's the thing, my original image I had done some tinkering ( I believe you helped me on v1engineering site)

So since I kept having issues with the UI I decided to start fresh and I seem to be having the same 5 minute time and then it crashes.

leitenky, 2019-02-13:
@jeffeb3 could you try unplugging your CNC from the pi and clicking connect? I am playing with my pi in the house not connected to the CNC because it's easier for me and that seems to be the cause of the memory leak that crashes CNCJs server

leitenky, 2019-02-14:
Hey @jeffeb3 just following up I've tried another fresh image of v1pi and tried to get through a 1.5 hour cut.  After about 45minutes CNCJs server seemed to stop working.  
Do you use a pi3b or pi3b+? I'm going to just order another SD card (even though it passes all the wrote tests I've thrown at it) just to fully rule the card out.  

I'll keep you updated.  Thanks for all your help!

jeffeb3, 2019-05-02:
I have it on a few pi versions, including a zero.

There were two CNC.js bugs related to Marlin in 0.9.20. I would suggest either trying the pre-release I just posted, or manually updating cnc.js.

Bretthansenpc, 2019-07-18:
i have the same issue 5 minutes and it runs out of memory and shuts down on a pi 3b+ im on the newest beta version too

jeffeb3, 2019-07-18:
Are you running the X server on the pi too, or just the CNC.js server?

Bretthansenpc, 2019-07-18:
just the way you send it right now no x server

ehansen2580, 2019-12-29:
Was this ever resolved?  I'm getting this same issue with the most recent release of the image

jeffeb3, 2019-12-31:
Do you have the cnc connected? The issue above was related to trying to connect, and not having a controller to connect to, so it was probably connecting via the builtin serial.