jeffeb3/v1pi #4: TinyWeb Console for 320x240 LCD Display

aperoap, 2019-02-24:
hello, i use your IMG v1pi on pi 3+. Is there a way to install TinyWeb Console for 320x240 LCD Display?

jeffeb3, 2019-02-24:
Yes. There are two parts to the problem.

1) installing the LCD, installing the x server, and Configuring the pi to show it's display there.
2) Installing the pendant code, configuring cnc.js to run it, and configuring a web browser to show it.

Have you already tried 1? Is there a good screen and instructions you followed?

aperoap, 2019-02-24:
I can not find the installation folder and cnc.js :)

jeffeb3, 2019-02-24:
npm installed it in `/usr/local/lib/node_modules/cncjs`

If you want to add the -m option to add the pendant module, the startup command is in `/etc/default/cncjs`

jeffeb3, 2019-02-24:
The default content in the /etc/default file is from here: https://github.com/jeffeb3/v1pi/blob/master/src/modules/cncjs/filesystem/root/etc/default/cncjs

And the command that installed cnc.js is here: https://github.com/jeffeb3/v1pi/blob/master/src/modules/cncjs/start_chroot_script#L24

If it works for you, I wouldn't mind trying it myself, and maybe including it in the next (TBD) version of the image. I'm assuming it's pretty small, so it's not too terrible to run all the time. I would just add the commands for installing the pendant code after the npm command in the start_chroot_script.

aperoap, 2019-02-25:
[image]
the server is started correctly, but the page gives an error.

[image]

do you have an idea what this may be?

how can i add the -m option in 
`/etc/default/cncjs`
?

jeffeb3, 2019-02-25:
You can add the -m <blah blah> to DAEMON_ARGS. 

It should be port 5080 when that comes up.

aperoap, 2019-02-25:
Update / Upgrade Pi 3+
`sudo apt update && sudo apt upgrade && sudo apt dist-upgrade-y`

Install important packages
`sudo apt-get install --no-install-recommends xserver-xorg`
`sudo apt-get install --no-install-recommends xinit`
`sudo apt-get install raspberrypi-ui-mods`

now config pi to boot with DESKTOP GUI (B1 Desktop  / CLI / Desktop Autologin Desktio GUI, automatically logged in as pi user.)
`sudo raspi-config`

for Chrome kiosk - as the :  https://obrienlabs.net/setup-raspberry-pi-kiosk-chromium/
`sudo apt-get update && sudo apt-get upgrade

sudo nano /etc/lightdm/lightdm.conf

[SeatDefaults]
 
autologin-user=pi

nano /home/pi/.config/autostart/kiosk.desktop
`
add
`[Desktop Entry]
Type=Application
Name=Kiosk
Exec=/home/pi/kiosk.sh
X-GNOME-Autostart-enabled=true`

Create kiosk script

`nano /home/pi/kiosk.sh`

and add

`#!/bin/bash

# Run this script in display 0 - the monitor
export DISPLAY=:0

# Hide the mouse from the display
unclutter &

# If Chrome crashes (usually due to rebooting), clear the crash flag so we don't have the annoying warning bar
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/pi/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"Normal"/' /home/pi/.config/chromium/Default/Preferences

# Run Chromium and open tabs
/usr/bin/chromium-browser --window-size=480,320 --kiosk --window-position=0,0 http://localhost:5080/cnclcd/ &

# Start the kiosk loop. This keystroke changes the Chromium tab 
# To have just anti-idle, use this line instead:
# xdotool keydown ctrl; xdotool keyup ctrl
# Otherwise, the ctrl+Tab is designed to switch tabs in Chrome
# #
while (true)
 do
  xdotool keydown ctrl+Tab; xdotool keyup ctrl+Tab;
  sleep 15
done
`
Download tinyweb

`wget https://github.com/cncjs/cncjs-pendant-tinyweb/releases/download/latest/cncjs-pendant-tinyweb-1.2.4-352767f.zip

unzip cncjs-pendant-tinyweb*.zip -d /home/pi/
rm -r cncjs-pendant-tinyweb*.zip

sudo mv /home/pi/cncjs-pendant-tinyweb* /home/tinyweb
`

`sudo nano /etc/default/cncjs
`
change  `DAEMON_ARGS="--host=$HOST --port=$PORT"` to `DAEMON_ARGS="--host=$HOST --port=$PORT -m /cnclcd:/home/tinyweb/src"`

now reboot pi

"sudo reboot now"

it works very well and is very helpful!

aperoap, 2019-02-25:
Update / Upgrade Pi 3+
`sudo apt update && sudo apt upgrade && sudo apt dist-upgrade-y`

Install important packages
`sudo apt-get install --no-install-recommends xserver-xorg`
`sudo apt-get install --no-install-recommends xinit`
`sudo apt-get install raspberrypi-ui-mods`

now config pi to boot with DESKTOP GUI (B1 Desktop  / CLI / Desktop Autologin Desktio GUI, automatically logged in as pi user.)
`sudo raspi-config`

for Chrome kiosk - as the :  https://obrienlabs.net/setup-raspberry-pi-kiosk-chromium/
`sudo apt-get update && sudo apt-get upgrade

sudo nano /etc/lightdm/lightdm.conf

[SeatDefaults]
 
autologin-user=pi

nano /home/pi/.config/autostart/kiosk.desktop
`
add
`[Desktop Entry]
Type=Application
Name=Kiosk
Exec=/home/pi/kiosk.sh
X-GNOME-Autostart-enabled=true`

Create kiosk script

`nano /home/pi/kiosk.sh`

and add

`
export DISPLAY=:0

sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/pi/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"Normal"/' /home/pi/.config/chromium/Default/Preferences

/usr/bin/chromium-browser --window-size=480,320 --kiosk --window-position=0,0 http://localhost:5080/cnclcd/ &

while (true)
 do
  xdotool keydown ctrl+Tab; xdotool keyup ctrl+Tab;
  sleep 15
done
`
Download tinyweb

`wget https://github.com/cncjs/cncjs-pendant-tinyweb/releases/download/latest/cncjs-pendant-tinyweb-1.2.4-352767f.zip

unzip cncjs-pendant-tinyweb*.zip -d /home/pi/
rm -r cncjs-pendant-tinyweb*.zip

sudo mv /home/pi/cncjs-pendant-tinyweb* /home/tinyweb
`

`sudo nano /etc/default/cncjs
`
change  `DAEMON_ARGS="--host=$HOST --port=$PORT"` to `DAEMON_ARGS="--host=$HOST --port=$PORT -m /cnclcd:/home/tinyweb/src"`

now reboot pi

"sudo reboot now"

it works very well and is very helpful!

aperoap, 2019-02-25:
Update / Upgrade Pi 3+
`sudo apt update && sudo apt upgrade && sudo apt dist-upgrade-y`

Install important packages
`sudo apt-get install --no-install-recommends xserver-xorg`
`sudo apt-get install --no-install-recommends xinit`
`sudo apt-get install raspberrypi-ui-mods`

now config pi to boot with DESKTOP GUI (B1 Desktop  / CLI / Desktop Autologin Desktio GUI, automatically logged in as pi user.)

`sudo raspi-config`

for Chrome kiosk - as the :  [here](https://obrienlabs.net/setup-raspberry-pi-kiosk-chromium/)

`sudo apt-get update && sudo apt-get upgrade`

`sudo nano /etc/lightdm/lightdm.conf`

[SeatDefaults]
 
autologin-user=pi

`nano /home/pi/.config/autostart/kiosk.desktop`
add

[Desktop Entry]
Type=Application
Name=Kiosk
Exec=/home/pi/kiosk.sh
X-GNOME-Autostart-enabled=true

Create kiosk script

`nano /home/pi/kiosk.sh`

and add

`export DISPLAY=:0

unclutter &

sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/pi/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"Normal"/' /home/pi/.config/chromium/Default/Preferences

/usr/bin/chromium-browser --window-size=480,320 --kiosk --window-position=0,0 http://localhost:5080/cnclcd/ &

while (true)
 do
  xdotool keydown ctrl+Tab; xdotool keyup ctrl+Tab;
  sleep 15
done`

Download tinyweb

`wget https://github.com/cncjs/cncjs-pendant-tinyweb/releases/download/latest/cncjs-pendant-tinyweb-1.2.4-352767f.zip`

`unzip cncjs-pendant-tinyweb*.zip -d /home/pi/`
`rm -r cncjs-pendant-tinyweb*.zip`

`sudo mv /home/pi/cncjs-pendant-tinyweb* /home/tinyweb`

`sudo nano /etc/default/cncjs`
change  `DAEMON_ARGS="--host=$HOST --port=$PORT"` to `DAEMON_ARGS="--host=$HOST --port=$PORT -m /cnclcd:/home/tinyweb/src"`

now reboot pi

`sudoreboot now` 

it works very well and is very helpful!