jeffeb3/v1pi #6: Add poweroff to cnc.js or landing page

jeffeb3, 2019-06-29:
You can't poweroff from cnc.js.

TheKyleJohnson90, 2019-10-06:
You can add these via the Settings -> Commands feature in cncjs.

Example: 
title: Reboot
commands: sudo reboot

heymish, 2021-12-11:
[image]

Just needs to add to the .cncrc. I will add a PR for this once the other PR is merged.
```
,
    "commands": [
    {
      "title": "Reboot",
      "commands": "sudo shutdown -r now"
    },
    {
      "title": "Shutdown",
      "commands": "sudo shutdown -h now"
    }
  ]
```

 It uses the Octopi commands which have been added to `/etc/sudoers.d `so it all just happily works.

I have tested this. It also works if the Pi user has a password set.