jeffeb3/v1pi #7: Issues on Raspberry Pi 3 A+

vector76, 2019-08-20:
I tried both 0.16.0 and 0.15.1 and neither worked on the Pi 3 A.

I'm fine with OctoPi, so you don't need to fix it on my account, just thought you would want to know.

jeffeb3, 2019-08-20:
What didn't work? I don't have any of the A models, unfortunately. 

I also noticed octoprint recommends using at least a 3B+ now, and I am still running 15.1 on a zero in my mp3dp (without a camera, but still).

vector76, 2019-08-20:
One version would show the landing page but wouldn't run either OctoPrint or CNC.js, and the other version wouldn't boot at all.  I plugged in a screen and it showed a big colored square and never changed from that.  I can't remember which version was the worse one, I'll try again and report back.

vector76, 2019-08-20:
2018-09-29-v1pi-no-hotspot.img is the one that hangs on a colorful square, I think the firmware is just out of date relative to the 3A+.

Now that I'm trying the other one (2018-09-29-v1pi-no-hotspot.img), it appears to be working!  I'm not sure what I did before, but this time I didn't touch anything except changing the wpa supplicant.txt on the boot partition, and it's working.  So nevermind about the problem!

closed 2019-08-20 by vector76