jeffeb3/v1pi #19: How to build with modded cnc.js version?

driedell, 2020-02-01:
I'm working on a fork of cncjs that sends marlin-compatible work offset zeroing commands instead of the "G10 L20 P1 X0" that it sends by default using these buttons.

[image]

I'd like to create a build of v1pi with this updated cncjs version bc I love having my Raspberry pi 4 drive both my MPCNC and my 3D Printer, but I was having trouble finding where you include cncjs in your source code.

Can you point me in the right direction? (I'm a bit of a software noob but I have used Docker before.)

jeffeb3, 2020-02-01:
I install it using npm here:

https://github.com/jeffeb3/v1pi/blob/master/src/modules/cncjs/start_chroot_script#L24

If you just want to edit your own version of CNC.js, then it probably makes sense to install it locally, in the /home/pi directory, and then have the service start your local version instead.

If you are interested in fixing this more generally, then I would suggest making the change in a pull request on cnc.js. If they agree with your change, and include it in the next version, then I can just rerun the script to rebuild the images. Then we test and release.

driedell, 2020-02-01:
Ah that's a good idea. After I install it locally, how do I make the v1pi service load my modded cncjs version over the default?

I know a bunch of people on the v1engineering forums and github have discussed various issues with marlin and cncjs. I'm hoping to come up with a few fixes and get them pushed out when they're ready.

LongLiveCHIEF, 2020-02-29:
Might make sense to have a volume mount for each of the cncjs and octoprint images or config file paths, so that you can mount in your own versions if needed. That's sorta the beauty of docker!