jeffeb3/v1pi #20: Enable Hardware Serial

neilferreri, 2020-03-06:
How would I enable access to the hardware serial?  I have a CNC hat from Protoneer, and it uses the RX & TX on the GPIO pins.
https://wiki.protoneer.co.nz/Raspberry_Pi_CNC

jeffeb3, 2020-03-06:
The serial port console needs to be turned off, and the serial port needs to be enabled. These can both be done in raspi-config. This article makes it pretty clear:

jeffeb3, 2020-03-06:
I didn't mean to close this, but I think this answered your question. If not, feel free to keep asking and we will get to the bottom of it.

closed 2020-03-06 by jeffeb3

neilferreri, 2020-03-06:
Thanks for the quick response. I've done all of that through the cli. The raspi-config probably makes it easier, though.
I'm wondering if I messed up my grbl installation. I'll report back.
Thanks again.

neilferreri, 2020-03-12:
Just wanted to clear this up.  The issue ended up being my CNC hat.  I ended up having to reflash a bootloader and then grbl.
Once that was working, I used the following commands to enable the serial port (maybe to help the next person):

`sudo sh -c "echo 'dtoverlay=pi3-miniuart-bt' >> /boot/config.txt"
sudo sh -c "echo 'enable_uart=1' >> /boot/config.txt"
sudo systemctl stop serial-getty@ttys0.service
sudo systemctl diable serial-getty@ttys0.service`

After a reboot, it workied