jeffeb3/v1pi #23: add more modules like cncjs

jiantaolovebingbing, 2020-06-12:
hi,
 if i  want to add my github project, how should i do to write the 'start_chroot_script'  file inside 'modules/projectName'  directory？
[image]
because i can't use npm to install the project , using git clone?

jeffeb3, 2020-06-12:
Those scripts get executed in an environment that emulates running them on a pi. You can basically run the same commands as you would as if you were installing it on the pi. When you finish, it will be saved in the image. You can get a lot more detail at custompios.

What project is this for? I'm curious.

jiantaolovebingbing, 2020-06-12:
I repeated these steps  on this project, but failed. 
[v1pi.log](https://github.com/jeffeb3/v1pi/files/4769149/v1pi.log)

But if our project is not on npm, how should I edit this part code?Or i have to publish a Npm package?
sorry, i'm not  familiar with bash code.

thanks
A project with elementary cnc and laser function

jeffeb3, 2020-06-12:
You can put git clone right in the file. It is kind of a pain to start with the scripts. I usually start with a fresh v1pi image installed and I use terminal commands to condigure what I want and I just keep a good record of the commands I used. Then the script ends up being just those commands.

It looks like maybe you're trying to install a node.js application. Does it run as a server with a web ui or does it need to have a desktop running?

Octoprint and cncjs are configured to run at startup, and then I do some haproxy condiguration so port 80 shows the landing page. The landing page forwards to the haproxy port of each service.

jeffeb3, 2020-06-12:
I think the errors you're getting might be due to not cloning the submodules in dependencies.

jiantaolovebingbing, 2020-06-13:
yes, i just don't know how to start with the script. Is there some document to use using Git clone but not npm？

it just run as a server with a web ui.

jeffeb3, 2020-06-13:
I strongly suggest you get a pi running with v1pi, and just try commands on the command line. It will be way easier than waiting 10+ minutes for each mistake, and trying to sift throigh feedback in that log file.

On the command line, if you make a mistake, the tool will give you instant feedback. You can run something like this and immediately see the help:

`git clone - - help`

But, chances are, your task is pretty simple. Something like this:

```
git clone https://github.com/user/repo
cd repo
npm install
```
Then, you'll need to work out the systemctl stuff to be able to make it run at startup. That is all going to be much more complicated. It will be really hard to figure that out without a live pi running.

jiantaolovebingbing, 2020-06-14:
ok, sorry, What i mean is not clear enough。
I know git and npm commands, i just don't know how to make it run at startup.
i will  close the issue,thanks

closed 2020-06-14 by jiantaolovebingbing