jeffeb3/v1pi #26: CNCjs not working at all.

CorvoDewittV2, 2020-08-19:
Hi I just set up Version 0.17, and Octoprint works fine, but CNCjs won't do anything. It connects, the console shows up after I select marlin and hit open, but it won't do anything. After about 2 minuets I get a pop up that says server has stopped working.  I will say it is my first rodeo with CNCjs but I have 2 printers running octoprint, so i kinda know my way around rpi but by no means an expert.

jeffeb3, 2020-08-19:
Do you have the right port?

Have you disconnected from octoprint?

CorvoDewittV2, 2020-08-19:
I was disconnected from octoprint, here is my order of operations:

Plug in pi and mpcnc
Wait 2 min to make sure everything is booted
Go to the server page 
Click on CNCjs
Connect using the only port listed, under marlin.
Console will pop up below the Connect tab
I can type commands into the console or press the move buttons but nothing happens (no ok's either)
2 min later the server crashes

jeffeb3, 2020-08-19:
Which port and which pi?

ttyAMA0 is the onboard serial, and not the usb. I usually have two options if I have Marlin connected.

CorvoDewittV2, 2020-08-19:
Oh huh ok I will have to check when I get home, I will report back with screenshots!

CorvoDewittV2, 2020-08-19:
[image]
[image]
This is as far as I get. After a little while it will just crash. It's actually not crashing right now...but still not moving the machine. Am I doing something dumb? I really hope I am doing something dumb.

CorvoDewittV2, 2020-08-19:
WAIT I think I fixed it by changing the baud rate to 250000! Yeah it works now. Issue was between the keyboard and chair. :)

closed 2020-08-19 by CorvoDewittV2

jeffeb3, 2020-08-19:
Glad it's working for you. Those goofy utf characters are a good sign of bad baud rate.