jeffeb3/v1pi #29: Is possible to set home manually?

elongo, 2020-09-06:
Hi. Does anyone knows if it
Is possible to set the home manually in octopi, to start the the cnc job? And how?

Thanks you!!

e

jeffeb3, 2020-09-06:
You can set up whatever macros you want.

I would set these up:

Set X Zero: G92 X0
Set Y Zero: G92 Y0
Set Z Zero: G92 Z0

You can also adjust the starting gcode to include these, and just set the tool at 0,0,0 to start.

Watch out for the M84 in the pause/ending gcode.

closed 2020-09-06 by jeffeb3

elongo, 2020-09-06:
Thank you. Good indications to have also.

Sorry, perhaps I didn’t explain myself enough! I was asking about the
option of, before starting the job, move the router to a specific location
and the in octoprint say “Set home” (which will be set at the place where
the router is at that given moment.

Something like in Repetier. Where you have the “Set Home” button. Sometimes
is very handy.

e

jeffeb3, 2020-09-06:
That's what I was suggesting, but one axis at a time:

Set all zero: G92 X0 Y0 Z0

elongo, 2020-09-07:
Awesome, man. Got it! Thanks..