jeffeb3/v1pi #33: Unable to upgrade Python3.

InoxDragoon, 2021-08-31:
There are a couple Python plugins that require Pythin 3.3 or higher in OctoPrint that I can't install because Python is at 2.7.16. All the google-fu I've done all state that OctoPrint should be reflashed in order to get Python to version 3+. However, if I do that, then I lose CNC.js. 

Are there plans to update this package to a newer version on OctoPi?

jeffeb3, 2021-08-31:
I have been really irresponsible with this image. I need to update it, install it, and test it. But it keeps staying at the bottom of my todo list. Sorry. I do hope to pick it back up at some point. But right now, this project is pretty much without any maintenance. Until I pick it back up, I suggest using octopi directly.

closed 2021-08-31 by jeffeb3

InoxDragoon, 2021-08-31:
Understood, thanks anyway!